OpenROV/openrov-software #339: 2.5.1 software having odd behavior with motor and setting updates

BrianAdams, 2015-02-18:
From Walt:

I ran my 2.7 a bit more today before packing it away for my trip.  The first couple of boots, there were various failures in the motors- they didn’t run backwards, they didn’t run at all.  Sometimes the MTARG values looked normal, other times going backwards the value would stay at 1500, then suddenly drop to 1000.  One time the motor direction checkbox on the starboard motor, which I had given up on being persistent, suddenly checked itself.

 On the 4th boot everything seemed fine, except for the issue with the motor direction checkbox.

BrianAdams, 2015-03-04:
Confirmed with Walt's 2.7 running 2.5.1-final. ATMega image was 1743c033e4014015b6660b9be2063e964cfc02e8 .  Which stayed the same when I reflashed it so that must be the latest.
When first starting the ROV forward works on all motors but reverse does not.  The Mtarg() shows the ATMEGA received the commands, but Motor() shows that the motors did not actually get sent the new servo positions.  The Motor Modifiers in the telemetry stream *show* as all populated.  My guess is that the Motor Modifiers is incorrect and that something has set the values after the UI was sent the update.  

Workaround Confirmed: Going to the diagnostics screen and closing it using the close arrow causes the motor modifiers to be resent.

codewithpassion, 2015-03-05:
That's a common issue I faced as well.
Only affects ROVs where you have to reverse one or more motors.

Before you work on this. Check out the reworked versions from my PR (new
ui) first.
I think it didn't happen there

BrianAdams, 2015-03-16:
A check in the logs shows an errant message being sent that screwing things up:
```
rov@OpenROV:/var/log$ cat openrov.log | grep mtrmod2
cmd: mtrmod2(200,-200,200)
*cmd: mtrmod2(ligt,12.5)*
cmd: mtrmod2(200,-200,200)
cmd: mtrmod2(200,-200,200)
cmd: mtrmod2(200,-200,200)
cmd: mtrmod2(200,-200,200)
cmd: mtrmod2(200,-200,200)
cmd: mtrmod2(200,-200,200)
cmd: mtrmod2(200,-200,200)
```

BrianAdams, 2015-03-17:
Merged to pre-release. ff249d8,

closed 2015-05-13 by BrianAdams