OpenROV/openrov-software #343: Proxy stop fails [30.0.0 Pre-release build 152]

BrianAdams, 2015-03-17:
rov@OpenROV:~$ sudo /etc/init.d/openrov-proxy stop
[sudo] password for rov:
/opt/openrov/proxy//app.js
[....] Stopping OpenROV NodeJS proxy server: openrov-proxy[....]   --->  OpenROV NodeJS proxy server stopped: openrov-proxyfatal: bad config file line 1 in /etc/gitconfig
fatal: bad config file line 1 in /etc/gitconfig
fatal: bad config file line 1 in /etc/gitconfig

BrianAdams, 2016-11-14:
Not appearing in latest dev image. Closing for now.

closed 2016-11-14 by BrianAdams

saulibanez, 2018-02-14:
good afternoon,
I have the same problem with the proxy, is there a solution?

rov@OpenROV:~$ sudo /etc/init.d/openrov-proxy stop
[sudo] password for rov: 
/opt/openrov/proxy//app.js
[....] Stopping OpenROV NodeJS proxy server: openrov-proxy[....]   --->  OpenROV NodeJS proxy server stopped: openrov-proxyfatal: bad config file line 1 in /etc/gitconfig
fatal: bad config file line 1 in /etc/gitconfig
fatal: bad config file line 1 in /etc/gitconfig
. ok 

Regards