OpenROV/openrov-software #351: Proposal: Consolidate all software issues lists

BrianAdams, 2015-03-23:
Concern
---
That we have a small team and can loose focus of the issues tied to the sub projects.  

Goal
----
By centralizing the issue repositories the issues will be better groomed. Users will clearly have a central location to report the issues without having to know which sub project might actually be responsible for the code.

Steps
----
Create tags for:
* Controllerboard-firmware
* Dashboard
* Cloud9
* etc...

Move all of the issues (recreate and close) from the other repositories.  Disable the issues on the other repositories.

As we check-in code in other repositories that fix issues, get good at using the user/repo#issue format to reference the issue in checkin comments.

@codewithpassion, any concerns with moving forward?

BrianAdams, 2015-03-23:
Phase 2 IMHO will be to go ahead and migrate the cockpit code in to a new repo called cockpit and replace the actual code with references and documentation to the rest of the software.

codewithpassion, 2015-03-23:
Cool with that.

BrianAdams, 2015-03-24:
Issue consolidation complete

closed 2015-03-24 by BrianAdams