OpenROV/openrov-software #356: Cloud9 IDE failing on 2.5.1-final image

BrianAdams, 2015-03-23:
http://192.168.7.2:3131/static/ace/build/ace.js : Failed to load resource: the server responded with a status of 404 (Not Found)

File is installed here:
/opt/cloud9/node_modules/ace/lib/ace

Suspect something moved since the config worked.

BrianAdams, 2015-03-24:
!@$!@$!!^!@# 
So, the solution is referenced https://stackoverflow.com/questions/26894708/cloud9-on-ubuntu-getting-ace-js-404-error-in-chrome, but the github repo closed the issue list so I can't see the solutions.

codewithpassion, 2015-03-24:
Maybe we should just find out where the BBB guys get cloud9 from and use
that.

There's a repo somewhere.

BrianAdams, 2015-03-24:
Yea, it is in purgatory with BBB as well. The old repo got nuked and the
new cloud9-core is in beta.

BrianAdams, 2015-03-25:
It appears that the first time you run the service it does  a make on file.  There seems to be a missing package.json dependency, packager.  With that missing the ace project does not get built.

BrianAdams, 2015-10-28:
Fixed with new cloud9

closed 2015-10-28 by BrianAdams