OpenROV/openrov-software #367: The commands that the firmware execute should immediately return updated state

BrianAdams, 2015-03-24:
Today many of the commands set the global state and wait for the 1 second update to send them out which adds a lot of lag to the system if someone is monitoring these values.  For instance, turning on the lights should immediately return the lights on value as well as keep tabs of overall state.

spiderkeys, 2015-09-26:
Bumping this on my radar. If I remember correctly, the firmware actually is responding immediately, it was the update loop in the Node process that was running at 1Hz. Will need to take a look at those loops sometime soon to get it more responsive.

BrianAdams, 2016-05-05:
@spiderkeys is this in the 30.0.4 arduino code?

spiderkeys, 2016-05-05:
Looks like Lights immediately return state, as do lasers, CPin commands in general, and autopilot commands.

closed 2016-07-25 by spiderkeys