OpenROV/openrov-software #369: Proposal: Change default keyboard layout to match typical FPS games

BrianAdams, 2015-03-24:
The goal for this change is:
1. Make it more intuitive on how to control the ROV
2. Move away from using the modifier keys (crtl.fn,shift) that when combined with other keys overlaps with reserved commands that we cannot override (such as Ctrl+Arrow Key in OSX that cause the screen to swap to another screen).

The proposed key combination would be:

Movement Controls:
====
Key | Command 
--- | ---
W | move forward 
S | move back 
A | strafe left (reserved)
D |  strafe right (reserved)
Q | roll left
E | roll right
Up | left up -or- pitch down
Down | lift down -or- pitch up
Left | yaw left
Right | yaw right
Shift-W | pitch down -or- lift up 
Shift-S | pitch up -or- lift down

Secondary Controls:
====
Key | Command 
--- | ---
I/K | camera tilt
J/L | reserved camera pan
O/U | reserved camera zoom
G | depth hold
H | heading hold
1-5 | motor power settings
6-7 | light dim/bright
8 | lasers on
9,0 | reserve for illumination controls

The -or- commands show where the keyboard flops to give priority to the primary movement axis used to execute overall vertical adjustments.  So a ROV that tends to pitch to change depth would have pitch controls as the primary on the up/down arrows while a ROV that uses dedicated thrusters would have those as primary.

This also allows for no change in the left hand keyboard commands when introducing a mouse.

For the secondary controls I have focused on trying to keep like controls spaced together in a way that supports simple recall.  
* Power associated things on the number key, thruster power on left, illumination controls on right
* Reserving another 2-axis key combination (I,J,K,L) for camera gimbal
* Place the holds next to each other, keying off of the H for a heading hold and putting depth hold right next to it in G.

*This needs to be driven extensively and tested. It is not meant to prevent customization, but is instead proposing the initial default keyboard layout to achieve the goals meant at the beginning of the document.*

codewithpassion, 2015-03-24:
While I agree with the general idea of looking into moving away from the
modifier keys, I'm not sure if the move to a left hand driven ROV makes
that much sense for most people.

The left hand control style comes from the fact that FPS games use a mouse
for control and we don't have a dedicated mouse control in mind I believe.

Given that, if I interpret the mapping correctly, I need now both hands on
the keyboard to move to ROV forward and up/down where's today, with most
keyboards having a ctr/shift close to the arrow keys, I can use one hand.

In any case, if it should be decided to go ahead, we have to be able to
give the user's an easy way to switch to the 'old style' controls.
The input controller is setup to accept different mappings so a profile
solution should be easy to add.

Ultimately, we might as well just implement the ui to assign your own
profiles when we do this change.

I strongly disagree with moving away from the current scheme with no way
back (and editing code is not a valid way to go back).

spiderkeys, 2017-01-10:
Here are my inputs for specifically the 2X series:

```
2X Presets
-------------
W	move forward
S	move back
A	yaw left
D	yaw right
Q	vertical down
E	vertical up

[ ] \	camera tilt down/up/center
, . /	lights down/up/toggle
l	lasers on
G	depth hold
H	heading hold
1-5	motor power settings
- =	motor power down/up
```

Rationale for camera servo and lights are that they are groups of related keys with a consistent layout.
[ = left hand side, camera down
] = right hand side, camera up
\ (paired with pipe character | ) = camera center
[ | ]

, (paired with <) = light down
. (paired with >) = light up
/ = light center (follows same pattern as camera keys, though no good visual pairing here like |)

nerdgilbert, 2017-01-10:
@BrianAdams Thoughts on the above?

BrianAdams, 2017-01-10:
I would suggest the C for vertical down instead, only because it moves a primary control to a more dominate finger and it is spatially below the up key which should make it more intuitive.

We ran in to an issue before where the symbol keys are not standard across international keyboards, so if we have groups of keys such as the down/up/toggle grouping, it may be better to move to actual keyboard keys for a default layout.

BrianAdams, 2017-01-10:
Another option would be to use C for up and Space for down which allows lift to be controlled by just the thumb .

nerdgilbert, 2017-01-10:
Didn't know about the international keyboard configuration. Perhaps we do the following

```
i o p     camera tilt down/up/center
b n m   lights down/up/toggle
```

BrianAdams, 2017-01-10:
Still looking for a good reference, but you can see examples here:
[image]