OpenROV/openrov-software #371: Need a way to remove the photos from the ROV disk

BrianAdams, 2015-03-24:
Currently you can take photos but the only way to remove them is to ssh in to the ROV and execute:
`sudo rm -rf /var/www/openrov/photos`

There should be a way to delete them without having to ssh on to the ROV

BrianAdams, 2016-08-11:
For anyone that wants to get the photos, you can use `scp` to copy them. Or you can simply right click on the image and download it to your computer.

Going forward, the images will always be captured on the browser so this is no longer an issue.

closed 2016-08-11 by BrianAdams

zjohns2, 2017-09-29:
Just ran through this with someone with a 2.8, had some good feedback.

⦁	Go to 192.168.254.1:3009  (“Wetty-The Web TTY Terminal Emulator” now shows with the numbers but I don’t have a clue how it got there)
⦁	First prompt type “rov” and press enter
⦁	At the Password prompt type “OpenROV” and press enter You won’t see the password being typed in.
⦁	Next prompt type “sudo rm -rf /var/www/openrov/photos”
⦁	At this Password prompt type “OpenROV” and press enter

At this point the photos have been deleted but there is no indication that they have been deleted nor did they ask “are you sure you want to delete your photos”. When you are deleting something this question is normally asked.

Also, you cannot now go back to the cockpit and proceed to copy new photos. You must first unplug everything from your computer, then reconnect to the ROV before you can copy any photos again.

It should look like this:
[image]