OpenROV/openrov-software #393: NewUI Refactor: Telemetry screen no longer allows scrolling

BrianAdams, 2015-04-13:
The system used to let you scroll down to see additional telemetry if it fills the screen

codewithpassion, 2015-05-04:
I still get the scrollbar 
[image]

BrianAdams, 2015-05-11:
Confirmed we are seeing this as an issue on OS/X + Chrome

covedesign, 2015-05-13:
I have the same problem with scrolling in OSX + Chrome

jacobrosenthal, 2015-06-20:
These scroll for me as of 65205a8
OSX 10.10.3 (14D136)
43.0.2357.124 (64-bit)

Can we get some versions?

BrianAdams, 2015-08-11:
One of the issues is that the size of the area is about 30px or so longer than shown. The footer covers the bottom part.  If there is not enough data to actually overflow the element the scrollbar won't show.  Even if you do scroll, you cannot get to the bottom covered by the footer.

BrianAdams, 2016-08-11:
This area was moved out of the new-ui cockpit.  The classic cockpit is being dialed in.  If it is still an issue in cockpit after the refactor we will reopen.

closed 2016-08-11 by BrianAdams