OpenROV/openrov-software #433: USB hub locks up

BrianAdams, 2015-07-17:
From syslog: 

>#musb_host_rx 1654: RX13 dma busy, csr 2020

I unplugged one camera, plugged in another and then I could no longer access it and got these messages in the logs.

spiderkeys, 2015-07-17:
From reading around and browsing through the musb_host source, it seems that when the first camera is unplugged, the USB host interface is left in some bad state and can't be used again until restarted.

```
1652         if (unlikely(dma_channel_status(dma) == MUSB_DMA_STATUS_BUSY)) {
1653                 /* SHOULD NEVER HAPPEN ... but at least DaVinci has done it */
1654                 ERR("RX%d dma busy, csr %04x\n", epnum, rx_csr);
1655                 goto finish;
1656         }
```

Seems this should never happen, haha. Not sure if there is a way to prevent that from happening or fix it, but there may be a way to detect it and do a hard reset on the USB host interface without requiring a BBB shutdown, but I haven't found a way to do the reset yet.

I'm curious to know if this problem happens when you connect through a USB hub.

BrianAdams, 2015-07-18:
Looks like I was also able to create it without swapping cameras.  Just several runs of bringing the camera feed up with avconv.

>Jul 17 22:04:26 openrov kernel: [61011.095498] musb_ep_program 896: broken !rx_reinit, ep13 csr 0003
Jul 17 22:04:26 openrov kernel: [61011.101790] musb_host_rx 1654: RX13 dma busy, csr 2020

BrianAdams, 2015-07-18:
Something to check. There are older kernel patches to force musb to fifo mode 5 but the dev image is currently using 4. 

https://github.com/Angstrom-distribution/meta-ti/blob/master/recipes-kernel/linux/linux-mainline-3.2/beagle/0011-default-to-fifo-mode-5-for-old-musb-beagles.patch

>rov@openrov:~$ dmesg | grep fifo
[    3.016039] musb-hdrc: setup fifo_mode 4
[    3.025448] musb-hdrc: setup fifo_mode 4