OpenROV/openrov-software #438: Lights will start flashing even with communication working

briangrau, 2015-07-30:
The lights on the lightboard will blink as if there was a loss of communication even with full communication. The video and controls all work. This is with software 30.0.2. Seen in the lab as well as by users.

BrianAdams, 2015-08-28:
There is some odd process that is kicking off and pausing non-essential services when the time on the beaglebone gets updated.  The symptoms of which are:

1. Red communication lights in lower right
2. Still have valid communication latency time in lower right
3. All telemetry stops updating
4. The lights start blinking on the ROV
5. The Cyclon lights on the beaglebone freeze.
6. Still have control of the motors.

After about 2 minutes the system finishes whatever tasks it was running and the system starts working as normal again.

A current work around is to disable the date/time update, but that has consequences in breaking ssl communication out for updates.

ianinbc, 2015-09-06:
Is there anything to report on this issue?

BrianAdams, 2015-09-07:
We know the issue is that the node.js setInterval loops stop executing for some duration and the start going again.  The issue spans across multiple node.js processes all at the same time, so it's not our code.  There are some known bugs in the 10.30 version of node.js which we may be triggering that were fixed in version 11 but never back ported. Specifically, if you set the date of your system back 5 minutes or so, the setInterval loops will essentially pause in the system... very much like we are seeing.

If there is a process running that is pushing the system date back a few minutes that could do it. We just have not found that process yet.  Perhaps it has something to do with the real-time linux extensions?

ianinbc, 2015-09-07:
Thanks. I wish I could help but linux is Greek to me.
							Ian

spiderkeys, 2015-09-26:
@BrianAdams @briangrau Are we still seeing this in 30.0.3?

BrianAdams, 2015-09-26:
I've not seen it so far... but then again I did not see it when developing
on 30.0.2 either.  That is one of the questions we are asking folks to
comment on, both from internal testing and from forum members downloading
the preview.  I have no reports either way so far.

briangrau, 2015-09-26:
This is still an issue. I have a 2.6 in the lab that has the lights flashing and the following conditions:

Green communication lights in lower right
Still have valid communication latency time in lower right
All telemetry updating
The lights start blinking on the ROV
Still have control of the motors.

Will continue with more testing on different units.

sansoy, 2015-10-07:
We just finished building ours this morning and seeing this issue.  We do have full control of the vehicle but the lights keep flashing.

BrianAdams, 2015-10-07:
This is fixed in the 30.0.3 pre-release: https://forum.openrov.com/t/beta-30-0-3-software-pre-release/3376

closed 2015-10-08 by BrianAdams

sansoy, 2015-10-08:
works awesome! thanks!  will report any issues after our dive today.

Olly999-1, 2016-03-28:
Hi,
It seems the problem still exists. 
I run a 2.8 with version 30.0.3 image on a windows 10 laptop.

My first ever dry trial went perfectly. No problems at all.
Since then, I have had to change the IP address several times to get communication. I now find that I have the exact symptoms described by Brian Adams, above.  Iget about two minutes of normal activity, followed by two minutes of problems and then back to normal for two minutes , continuously going through this cycle.

briangrau, 2016-03-29:
More details from the above description can be found on the forum

https://forum.openrov.com/t/dive-lights-flashing-no-other-problems/3233/6