OpenROV/openrov-software #440: Which Kernel should we be using for the DevImage right now

BrianAdams, 2015-08-06:
The options appear to be

3.14 : Lots of reports of rock solid stability. Currently in the dev image
3.15 : What is being used for the stable release right now
3.19 : Seemed to be active before beaglebone team moved to 4.0.1+
4.1.x : Most recent

There are lots of trade-offs.  The active development from the BeagleBone team seems to have shifted to the 4.1.x kernel.  There are lots of bug fixes and @spiderkeys mentioned even better start up performance that the 3.14.41-ti-r63 kernel.

The 3.14 kernel that I have been testing in the dev image has been showing stability issues regarding the USB camera.  Note, upgrading to 4.1.4-bone15 does not seem to fix it. The symptom changes, but I have random device busy issue that won't clear.

There is an ongoing issue with the 4.1.x Kernel that is high priority where there are random reboots: https://groups.google.com/d/msg/beagleboard/6H0bWGi8JPY/sneyNhffkDEJ

So our options appear to be:

1. Wait for the 4.1.x to fix the reboot error that has been seen.  Find and fix the video stability issue. Release on 4.1.x

2. Fix the video stability issue and go forward with the 3.14 kernel

3. Rever to the 3.15 kernel we used before and take the his in the startup performance improvements.

spiderkeys, 2015-08-22:
My vote is for stability, for the time being. If 3.15 is rock solid in terms of stability, it is worth sacrificing the startup performance improvements until stability is achieved in the newer kernel versions. Are there any other downsides to using 3.15 besides startup time?

BrianAdams, 2015-09-23:
So I have gone ahead and pushed through to the 4.1.x series for our "stable" release set. Turns out some the Kernel lines are better supported than others.  Some of the Kernels will be adopted for long term support, LTS, where we can count on constant back ports of fixes.  If we "lock" to a kernel it really should be a LTS kernel.

To @spiderkeys question, yes, the 4.1.x series support the capemgr again for making changes to pin assignments via use space commands.  Big benefit which is why we can count on more support from RCN and team on the 4.1.x kernel than any of the 3.14-3.19 kernels.

BrianAdams, 2015-10-08:
Well. It's going to stay with the latest 4.1 for now.

closed 2015-10-08 by BrianAdams