OpenROV/openrov-software #454: Servo Tilt Software Limits Farther Than Mechanical Limits.

briangrau, 2015-08-31:
In practice the servo has 7 set positions in each direction. If you press the "q" more than 7 times the servo does not move any more. If you then press the "z" key the servo does not move until you negate the extra keystrokes you had from before.

Example:
Servo starts in middle position
Press "q" 7 times
Servo moves up 7 places
Press "q" 2 more times
Servo does not move (which it should not because of the mechanical limits in the e-chassis
Want to move servo down
Press "z" key 2 times
Servo does not move because it it bringing it back into the range of acceptable values.
Press "z" key and the servo starts to move.

The limits should be set so no matter the number of keystrokes, you can change direction and see movement.

BrianAdams, 2015-09-01:
After reviewing the code, a simple fix would be to change the ArduinoPhysics so that instead of using the limit() function to cut off values greater than 70 percent of the range, you could instead just do the map function but limit the range it is mapping to to be 70% of the theoretical value.

https://github.com/OpenROV/openrov-cockpit/blob/8ef7a40be2b9ec87829b80d6ed98ee9ecfc4aaa7/src/lib/ArduinoPhysics.js#L37

BrianAdams, 2016-05-05:
This has been fixed via settings in the settings panel

closed 2016-05-05 by BrianAdams