OpenROV/openrov-software #455: IMU2 fails to consistently initialize on 2.7 boards and earlier

BrianAdams, 2015-08-31:
https://openrov.ontestpad.com/project/1/browse?auth=8a44ada22177f7901ed8adac80c95c27

It appears that the BNO-055 is not happy being software reset more than once.  We have also seen two camps of behavior in the sensor, some that are okay with multiple software resets and some that are not.

BrianAdams, 2015-08-31:
The BNO055 is supposed to startup in configuration mode.  We have checks and put it in configuration mode if it is not.  I wonder if we could use that as an indicator that the sensor has already gone through our initialization routine... and thus not issue another soft-reset to the sensor.

BrianAdams, 2015-08-31:
@briangrau is the tester for this ticket

BrianAdams, 2015-09-04:
The current work fix includes getting rid of all of the Arduino resets.  The rc.local kicks off a test to figure out what type of board is connected.  That test causes the Arduino to reset.   We need to test, but I believe we can safely defer the controller board type test as the scripts that need it call it if it has not been called before.

Then again, we could also add logic so that when doing the software reset of the Arduino, we could find a way to signal the Arduino first so that it puts the I2c devices in "safe" mode before the actual reset takes place.  This would be better as it prevents folks from having to do power cycles after Arduino resets due to I2C devices locking up.

spiderkeys, 2015-09-26:
This bug also seems to be fixed in the latest image. @briangrau also reported that the sensor is now working on the 2.6 as well. Closing since this seems resolved.

closed 2015-09-26 by spiderkeys