OpenROV/openrov-software #459: Moving the camera causes 100% cpu

BrianAdams, 2015-09-04:
On one of the units, we managed to grab some logs that showed the system went in to "[sched_delayed] sched: RT throttling activated" which explains the non-critical processes freezing up as the scheduler has stopped executing them to focus cpu on a real-time process that is struggling.  

In the same dump we see "[  366.784925] CPU: 0 PID: 2295 Comm: mjpg_streamer Not tainted 3.15.5-bone4 #1".  Still looking that up.

The system also reported a write error to the usb device: "Aug 19 23:04:19 OpenROV udevd[2428]: error opening ATTR{/sys/devices/ocp.3/47400000.usb/47401c00.usb/musb-hdrc.1.auto/usb2/2-1/2-1:1.2/sound/card0/controlC0/../uevent} for writing: No such file or directory", but there were many other errors as well. Ultimately the system appears to dump and then re-aquire the usb camera which may be why the freeze eventually stops.

Initial searching seems to imply that we may need to look at a Kernel upgrade.  Nothing that specifically matches.  Not clear why we are seeing this just now.

BrianAdams, 2015-09-04:
Turns out the custom usb cable was faulty. Switching cables fixed the issue.  @briangrau care to comment on any followup needed?

closed 2015-09-04 by BrianAdams

briangrau, 2015-09-26:
Had another unit with a faulty usb cable after use...appears to be pinched and might be due to mechanical fatigue.