OpenROV/openrov-software #462: 30.0.3 pre-release samba cannot connect

BrianAdams, 2015-09-23:
Using OS/X

The systemd made them make changes to the samba system.  The smbd and nmbd services are now separately managed.  The single samba wrapper no longer works.

Tried connecting with the default config and with the old config that we used before.

The system also starts the active directory integrated auth by default now. Not sure if that is a factor.

The system also does not create logs unless you manually create the /var/log/samba folder.

BrianAdams, 2015-09-26:
@codewithpassion 
Here are the steps I have run through
Login to the image
append the following to /etc/samba/smb.conf
```
[OpenROV]
comment = OpenROV Cockpit
path = /
force user = rov
force group = admin
read only = No
guest ok = Yes
```

execute: `sudo systemctl restart smbd nmbd`
I check and it appears to create the \var\log\samba folder

When attempting to connect from OS/X I always get a connectivity error

BrianAdams, 2015-09-26:
@spiderkeys can you try connecting to the 30.0.3 image using your windows box via file share \\192.168.254.1\OpenROV

spiderkeys, 2015-09-26:
No bueno.

BrianAdams, 2015-10-08:
Passed QA for 30.0.3 for connecting via guest.  Still something wonky connecting with credentials. Moving to next release.

BrianAdams, 2016-12-28:
Have removed Samba as a feature.

closed 2016-12-28 by BrianAdams