OpenROV/openrov-software #473: 30.0.3 Changing motor directions does not consistently reconfigure the motor

BrianAdams, 2015-10-13:
When initially switching the direction of the starboard thruster and vertical thruster, I noticed that the starboard thruster was switched correctly, but the vertical thruster ended up moving the same direction for both shift and control commands.  If I opened and closed the diagnostics window a second time (without changing any values) the problem was fixed and the motor direction worked correctly.

BrianAdams, 2015-10-13:
The system makes several calls back to the ROV to reset its motor configuration.  If one of those calls fails (serial corruption), that could explain the above behavior.  Opening and closing the diagnostics windows resends the configuration back to the motors.

briangrau, 2015-10-13:
Noted in https://openrov.ontestpad.com/script/7/#81/7/

BrianAdams, 2016-12-28:
This has been fixed.

closed 2016-12-28 by BrianAdams