OpenROV/openrov-software #500: Move to using the visionmedia/debug module for client and server logging

BrianAdams, 2016-02-16:
This is a proposal to decide on a single debug/logging module to simplify the enabling and disabling of trace messages in the software stack.

https://github.com/visionmedia/debug

This is used by Socket.IO. Allows enabling logging by module at runtime.  

The goal would be to replace all of the console.log message sprinkled through the platform with much more descriptive debug(msg) that are disabled by default but can be enabled by during runtime as needed.  This should remove a lot of the noise that gets sent to the console making it easier to spot real problems, and it should make it easier to trace problems in the running system as the trace messages used in development are still available to be used for debugging a live system.

This topic will be kept open for a week for comment 27-Feb-2016.  +1, LGTM, and the like are welcome.

spiderkeys, 2016-05-17:
Looks good! Will start instrumenting this and test it out.

spiderkeys, 2016-05-17:
Todo: Wrap debugger to easily expose capabilities like splitting up console.log and console.error, setting app wide configurations, and making it easy to enable/disable logging.