OpenROV/openrov-software #505: MJPEG stream not coming up on recent dev image

BrianAdams, 2016-04-28:
Reported by Jim:
```
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: starting application
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: MJPG Streamer Version: svn rev:
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: Using V4L2 device.: /dev/video0
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: Desired Resolution: 1920 x 1080
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: Frames Per Second.: 30
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: Format............: MJPEG
Apr 28 20:51:40 openrov mjpg_streamer[9699]: MJPG-streamer [9699]: init_VideoIn failed
```

BrianAdams, 2016-05-15:
Changing Kernel again fixed any issues I happened to see.  Still have to finish cleanup on the Tahoe refactor to upgrade the MJPEG server to match the new announcement api.

spiderkeys, 2016-05-16:
MJPEG plugin also had a mistake where the video-registration event was being emitted on deps.rov instead of deps.globalEventLoop when run with an actual camera. Being fixed in branch: https://github.com/OpenROV/openrov-cockpit/tree/refactor_controller

BrianAdams, 2016-06-15:
Latest version seems to be trying to use " Transfer Type            Isochronous" instead of bulk.  Too much video causes the entire camera to freeze up.

BrianAdams, 2016-06-15:
Hey @jtrezzo here is where I am tracking the MJPEG issues on the DevImage

BrianAdams, 2016-06-15:
So the earlier versions that work are also using isoc. So this looks like another Kernel version issue :-(

BrianAdams, 2016-06-15:
K. We changed some defaults to move the camera to 1080p@30, if you have too large a JPEG image, the USB is not keeping up using ISOC.  Dialing back to 720p@30 seems to work fine.  A different side affect of the MJPEG format is that nginx is running about 35% CPU, presumably encrypting the outbound stream.

spiderkeys, 2016-07-25:
This issue is now resolved by the new MJPEG video plugin that @codewithpassion has put together.

BrianAdams, 2016-09-21:
The nginx issue is fixed.  The underlying issue with the Genius camera throwing too many interrupts is still open.