OpenROV/openrov-software #513: File downloads through proxy using wget or curl fail

spiderkeys, 2016-07-19:
Currently seen on both RPI and BBB builds, wget and curl downloads will stop at 99%. git operations still seem to work fine.

Examples that will fail:
```bash
wget http://openrov-software-nightlies.s3-us-west-2.amazonaws.com/jessie/geomuxpp/openrov-geomuxpp_1.0.0-1~14_armhf.deb
```

```bash
curl http://openrov-software-nightlies.s3-us-west-2.amazonaws.com/jessie/geomuxpp/openrov-geomuxpp_1.0.0-1~14_armhf.deb --output geomuxp.deb
```

The curl example will eventually time out and print:

```bash
curl: (18) transfer closed with 31648 bytes remaining to read
```

BrianAdams, 2016-11-14:
Latest dev image is showing a problem where it appears the browser is passing back data to the service on the ROV that binary.js is unable to interpret.  Perhaps related to the node 6 upgrade?

BrianAdams, 2016-11-14:
Disabling in the dev image until we stabilize the behavior