OpenROV/openrov-software #515: Latest Dev Image has ttyO0 on boot, but no ttyO1 on BBB

BrianAdams, 2016-07-26:


BrianAdams, 2016-07-26:
Deleting the unused section in the .dts file seems to fix the issue.

spiderkeys, 2016-07-26:
Fixed in: https://github.com/OpenROV/openrov-cockpit/commit/4b516c429950f7c7c8d446ddfac73e1940bb7166

Also added some logging statements to orov-sysdetect to see whether or not the system gets automatically rebooted, and if not, why.

spiderkeys, 2016-07-26:
Reboot problem was hit or miss. On the RPi, it seemed to always work, but on the BBB, sometimes the "process.exit" would trigger before the exec( reboot) and the reboot wouldn't happen. I promisified the exec so it is guaranteed to resolve/reject before proceeding to the process.exit clause. This should fix all of the issues noted here.

closed 2016-08-11 by BrianAdams