OpenROV/openrov-software #529: The Platform-abstraction layer nginx configs should live in a board neutral location

BrianAdams, 2016-08-02:
The dependency for the nginx configs is that nginx is used as the reverse proxy. It is an identical config no matter what board we run on.  

Given this issue as an example of where this will go wrong: https://github.com/OpenROV/openrov-cockpit/commit/aaf5c11cb8e1af9ff30263a5c5905605ef5a2d83

Recommend we move the nginx configs, and anything else that is not board dependent outside of the board specific installation.

@spiderkeys thoughts?

spiderkeys, 2016-08-02:
Sure, we can add a new install step in platform-manager/install/ in a shared image-customization step and put files like nginx.location in there.

spiderkeys, 2016-08-02:
Done in:

https://github.com/OpenROV/openrov-cockpit/commit/d715f23ab07f4c1cc834aa279108e0d34d79f6b5
https://github.com/OpenROV/openrov-cockpit/commit/06a5b5b62ad32c781227cedc194c42b0db8fa31c

Added a shared install stage and added that step to platform-manager/Installer.js. No modifications to image-customization required.

BrianAdams, 2016-08-02:
Looks good

closed 2016-08-02 by BrianAdams