OpenROV/openrov-software #534: External light not working

WaltHolm, 2016-08-03:
Conditions:  31.0.1.dev-pal-04 PC with Windows7, Chrome 51.0.2704.103m.  OROV 2.8 with camera upgrade.

When working in the new cockpit, was not able to activate the external lights.  I tried both of the external light controls, and neither activated the lights.

briangrau, 2016-08-11:
Needs testing

BrianAdams, 2016-08-11:
should be fixed in latest build

WaltHolm, 2016-08-13:
Tested in 31.0.1 Dev Pal 07 with Chrome 52.  Keypad now works, but mouse clicks to the XL indicators does not.

WaltHolm, 2016-08-13:
Whoops, pressed the wrong button.  Re-opening issue.

BrianAdams, 2016-08-13:
@spiderkeys said he knew off the top of his head what the issue was. Assigning to at least get the brain dump.

spiderkeys, 2016-08-15:
Fixed in: https://github.com/OpenROV/openrov-cockpit/commit/30cad62c6033286980225f4d7581fadda49edaf8

closed 2016-08-15 by spiderkeys