OpenROV/openrov-software #557: Prevent platform-manager from crashing cockpit when run without controller board

BrianAdams, 2016-09-09:
The platform manager should gracefully fallback to unknown board type instead of crashing.  This is to support developers that have a naked Raspi or BBB but no controller board.  Cockpit should start up and indicate that no controller board was found.

>Error starting plugins: Error loading critical plugin section: Plugin:platform-manager is
Sep 07 02:05:43 openrov orov-cockpit[738]: Stack trace: Error: Error loading critical plugin section: Plugin:platform-manager is inva
Sep 07 02:05:43 openrov orov-cockpit[738]:     at /opt/openrov/cockpit/src/lib/PluginLoader.js:151:15
Sep 07 02:05:43 openrov orov-cockpit[738]:     at tryCatcher (/opt/openrov/cockpit/node_modules/bluebird/js/release/util.js:16:23)
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Promise._settlePromiseFromHandler (/opt/openrov/cockpit/node_modules/bluebird/js/re
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Promise._settlePromise (/opt/openrov/cockpit/node_modules/bluebird/js/release/promi
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Promise._settlePromise0 (/opt/openrov/cockpit/node_modules/bluebird/js/release/prom
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Promise._settlePromises (/opt/openrov/cockpit/node_modules/bluebird/js/release/prom
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Async._drainQueue (/opt/openrov/cockpit/node_modules/bluebird/js/release/async.js:1
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Async._drainQueues (/opt/openrov/cockpit/node_modules/bluebird/js/release/async.js:
Sep 07 02:05:43 openrov orov-cockpit[738]:     at Immediate.Async.drainQueues [as _onImmediate] (/opt/openrov/cockpit/node_modules/bl
Sep 07 02:05:43 openrov orov-cockpit[738]:     at tryOnImmediate (timers.js:534:15)
Sep 07 02:05:43 openrov orov-cockpit[738]:     at processImmediate [as _immediateCallback] (timers.js:514:5)

spiderkeys, 2016-09-09:
Some of the error messages got cut off here. The "plugin: platform-manager is inva" parts. Also, what platform were you using when this happened? Mock, RPI, BBB?

When I wrote the board loading section, I specifically set it up to swallow the error and move on without any problems. The thing that causes the platform-manager setup process to fail hard is a failure in the CPU loading process. See the following code:

```js
  // Load interfaces
  return Promise.try(function () 
  {
    return LoadPlatformName(self.platform);
  })
  .then(LoadCPUInterface)
  .then(LoadBoardInterface)
  .then(function (platform)
  {
    console.log('PLATFORM: Successfully loaded configuration for a supported platform.');
    deps.globalEventLoop.emit('platform.supported');
    return self;
  })
  .catch(function (err) {
    //deps.globalEventLoop.emit( "platform.unsupported", error );
    console.log('PLATFORM: Failed to load platform details for this system: ' + err.message);
    return null;
  });

function LoadCPUInterface(platform) 
{
  console.log('PLATFORM: Loading CPU interface...');

  var CPUInterfaceLoader = require('./platforms/' + platform.name + '/cpu.js');

  return CPUInterfaceLoader
          .Compose(platform)
          .catch(function (err) 
          {
            console.log('Failed to load CPU interface: ' + err.message);
            throw err;
          })
          .then(function () 
          {
            return platform;
          });
}

function LoadBoardInterface(platform) 
{
  console.log('PLATFORM: Loading Board interface...');

  // TODO: Move this into Promise.try()?
  var BoardInterfaceLoader = require('./platforms/' + platform.name + '/board.js');

  return BoardInterfaceLoader
          .Compose(platform)
          .catch(function (err) 
          {
            console.log('Failed to load board interface: ' + err.message);  // Continue anyway. Board is optional to operation of cockpit
          })
          .then(function () 
          {
            return platform;
          });
}
```

Notice that the Board Loader does a catch without rethrowing and proceeds with a then to allow the platform manager to continue loading without issue. The CPU Loader rethrows the error on the catch to fail the loading process of the platform-manager.

I don't see anything that would cause failure other than not having the platform set up correctly (a platform.conf file needs to be installed in /opt/openrov/system/conf or the env variable process.env.PLATFORM needs to be set to the name of a supported platform). If there is no valid platform detected, then the require() calls will fail. I currently have a todo that moves the require step inside the promise to catch these failures explicitly, but the ultimate reason is that the platform wasn't configured correctly in either the image creation phase or in the environment variables for running mock.

spiderkeys, 2016-09-09:
Turned out the BBB being used was the A6A model which wasn't in the supported revisionInfo.json list. I have added it to master.

closed 2016-09-09 by spiderkeys