OpenROV/openrov-software #560: Crash in plugin-finder's info event

spiderkeys, 2016-09-30:
Happened when opening the settings menu (seemingly). Seems to be bower related.
 Any ideas @badevguru? 
```
Sep 30 00:42:14 openrov orov-cockpit[30988]: Got device update
Sep 30 00:42:14 openrov orov-cockpit[30988]: performing list for plugins
Sep 30 00:42:14 openrov orov-cockpit[30988]: performing list for plugins
Sep 30 00:42:16 openrov orov-cockpit[30988]: performing list for plugins
Sep 30 00:42:16 openrov orov-cockpit[30988]: performing list for plugins
Sep 30 00:42:16 openrov orov-cockpit[30988]: /opt/openrov/cockpit/node_modules/bower/lib/node_modules/q/q.js:155
Sep 30 00:42:16 openrov orov-cockpit[30988]:                 throw e;
Sep 30 00:42:16 openrov orov-cockpit[30988]:                 ^
Sep 30 00:42:16 openrov orov-cockpit[30988]: Error: Failed to execute "git ls-remote --tags --heads https://github.com/BrianAdams/openrov-plugin-communityexample.git", exit code of #128
Sep 30 00:42:16 openrov orov-cockpit[30988]: fatal: unable to access 'https://github.com/BrianAdams/openrov-plugin-communityexample.git/': GnuTLS recv error (-24): Decryption has failed.
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at createError (/opt/openrov/cockpit/node_modules/bower/lib/util/createError.js:4:15)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at ChildProcess.<anonymous> (/opt/openrov/cockpit/node_modules/bower/lib/util/cmd.js:102:21)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at emitTwo (events.js:106:13)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at ChildProcess.emit (events.js:191:7)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at maybeClose (internal/child_process.js:877:16)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at Socket.<anonymous> (internal/child_process.js:334:11)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at emitOne (events.js:96:13)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at Socket.emit (events.js:188:7)
Sep 30 00:42:16 openrov orov-cockpit[30988]:     at Pipe._handle.close [as _onclose] (net.js:493:12)
Sep 30 00:42:16 openrov systemd[1]: orov-cockpit.service: Main process exited, code=exited, status=1/FAILURE
Sep 30 00:42:16 openrov systemd[1]: orov-cockpit.service: Unit entered failed state.
Sep 30 00:42:16 openrov systemd[1]: orov-cockpit.service: Failed with result 'exit-code'.
```

BrianAdams, 2016-09-30:
The bower library we are using when running the info command goes to github to get the data. In the case the system in unavailable we get the error message. That was not being caught.  I'm now trapping that error.  
The bower library is using promises. It would be nice to inject the bluebird promises in to it so that we can use the retry feature anytime we touch outside services and only fail after 3 tries or so.

spiderkeys, 2016-10-01:
Another crash that happened at the same spot which looks like a git clone, which supposedly shouldn't have happened:

```
Sep 30 02:53:02 openrov orov-cockpit[14722]: /opt/openrov/cockpit/node_modules/bower/lib/node_modules/q/q.js:155
Sep 30 02:53:02 openrov orov-cockpit[14722]:                 throw e;
Sep 30 02:53:02 openrov orov-cockpit[14722]:                 ^
Sep 30 02:53:02 openrov orov-cockpit[14722]: Error: Failed to execute "git clone https://github.com/OpenROV/openrov-telerobotic_controller.git -b master --progress . --depth 1", exit code of #128
Sep 30 02:53:02 openrov orov-cockpit[14722]: Cloning into '.'...
Sep 30 02:53:02 openrov orov-cockpit[14722]: fatal: unable to access 'https://github.com/OpenROV/openrov-telerobotic_controller.git/': Failed to connect to localhost port 3000: Connection refused
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at createError (/opt/openrov/cockpit/node_modules/bower/lib/util/createError.js:4:15)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at ChildProcess.<anonymous> (/opt/openrov/cockpit/node_modules/bower/lib/util/cmd.js:102:21)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at emitTwo (events.js:106:13)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at ChildProcess.emit (events.js:191:7)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at maybeClose (internal/child_process.js:877:16)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at Socket.<anonymous> (internal/child_process.js:334:11)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at emitOne (events.js:96:13)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at Socket.emit (events.js:188:7)
Sep 30 02:53:02 openrov orov-cockpit[14722]:     at Pipe._handle.close [as _onclose] (net.js:493:12)
Sep 30 02:53:02 openrov systemd[1]: orov-cockpit.service: Main process exited, code=exited, status=1/FAILURE
Sep 30 02:53:02 openrov systemd[1]: orov-cockpit.service: Unit entered failed state.
Sep 30 02:53:02 openrov systemd[1]: orov-cockpit.service: Failed with result 'exit-code'.

```

BrianAdams, 2016-10-02:
Odd. It may be that the bower info command actually does a git clone and parses the bower.json file to get the details it needs.  Catching the error should still mask the issue. Will look in to using the github api to directly interrogate and parse the bower.json file if indeed that is all bower is doing.

nerdgilbert, 2016-10-18:
Got this issue multiple times today during normal operation of cockpit

nerdgilbert, 2016-10-18:
Last lines of node debug
`performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
performing list for plugins
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
Tue, 18 Oct 2016 22:56:58 GMT bridge Unsupported command: ping
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
sending plugins list to browser
Tue, 18 Oct 2016 22:56:59 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:00 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:01 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:02 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:03 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:04 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:05 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:06 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:07 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:08 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:09 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:10 GMT bridge Unsupported command: ping
Tue, 18 Oct 2016 22:57:11 GMT bridge Unsupported command: ping`

Thrown error related to openrov telerobotic plugin