OpenROV/openrov-software #569: Reports of video not showing up in cockpit upon start up of the ROV

nerdgilbert, 2016-10-11:
Seems to be a rare event. Need to verify the cause and conditions of the bug.

briangrau, 2016-11-08:
For the Pro Camera-HD on the #169 release it seems that it will not come up on the first time, it takes a fully system restart in order for the video to come up after any changes to the system (such as firmware or applying the software patch). This has been confirmed by a few users.

BrianAdams, 2017-01-21:
Script to reset the image back to stock configuration
```shell
#!/bin/bash

mv /lib/modules/4.4.30-ti-r65/updates/uvcvideo.ko .
mxcam reset || true
modprobe -r uvcvideo
depmod -a
touch /.need-depmod
cp uvcvideo.ko /lib/modules/4.4.30-ti-r65/updates

```

BrianAdams, 2017-01-21:
We had a listener that was registered after the socket was already open to accept connections, so if the connection came before the listener was registered... we were missing it.

closed 2017-01-21 by BrianAdams