OpenROV/openrov-software #574: The notification database access crashes

BrianAdams, 2016-10-18:
Fatal cockpit crash when changing a keybinding. Can probably be triggered by other actions as well.

Steps to reproduce:
1. Run cockpit in mock mode in VScode
2. Open cockpit in browser
3. Open Input manager
4. Change a keybinding and click accept.
5. Immediately after clicking accept the following fatal error will occur:

[image]

Probably related to file permissions, as /tmp/ belongs to root:root by default on linux.

BrianAdams, 2016-10-18:
Possible related issues:
Trying to open the database more than once: 
* https://github.com/louischatriot/nedb/issues/320
* https://github.com/louischatriot/nedb/issues/136

nerdgilbert, 2016-10-18:
Verified on a fresh install. Made pull with master and restarted computer. Ran cockpit in debug mode and immediately threw this error. 
[image]

BrianAdams, 2016-10-18:
Fix will go in soon. Workaround is in initdb (), in the promise try, check
if self.db exists, and return if it does.

BrianAdams, 2016-10-18:
```
    initDB() {                   
        var self = this;
        return bluebird.try(function() {
            if (self.db){
                //Only initialize once, changes to settings used by nedb will requires a process restart.
                return;
            }
            if (process.env.NODE_ENV == "development") {
```

closed 2016-10-18 by BrianAdams