OpenROV/openrov-software #577: Switch panel does not clean up elements on some applet switches

spiderkeys, 2016-10-28:
Steps to recreate:

1. Open cockpit, let it load fully (finish any firmware updates)
2. Open diagnostics panel
3. Issue an MCU Reset
4. Software Update window should appear in cockpit screen, When the Software Update window disappears, there will now be an additional copy of all switches. These can continue to stack on each firmware update operation.

Refreshing cockpit will clear the copies out.

spiderkeys, 2016-10-28:
Problem seems to be intermittent, can't recreate it every time.

BrianAdams, 2016-11-22:
Best guess is that somehow, the widgetRegistry is executing its logic a second time and wiring in the duplicate elements in to the container of the switches.  The switch panel itself is rather simple and simply displays a "switch" for each child element that has been added as content.

BrianAdams, 2016-11-22:
Confirmed the duplicates as child content elements.
[image]

spiderkeys, 2017-01-06:
Saw that this is still an issue in the current dev image.