OpenROV/openrov-software #579: Firmware failed to update on first boot with latest dev image

BrianAdams, 2016-11-22:
Initial boot with Dev Image debian-8.6-orov_dev-armhf-2016-11-22

Came up in the notification panel after first boot.

>Nov 22 07:20:44 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: "avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/opt/openrov/arduino/hardware/openrov/avr/cores/arduino" "-I/opt/openrov/arduino/hardware/openrov/avr/variants/mega" "-I/opt/openrov/firmware/libraries/Servo/src" "-I/opt/openrov/firmware/libraries/MPU9150" "-I/opt/openrov/firmware/libraries/MS5803_14BA" "-I/opt/openrov/firmware/libraries/MS5837_30BA" "-I/opt/openrov/arduino/hardware/openrov/avr/libraries/EEPROM" "-I/opt/openrov/firmware/libraries/ORUtil" "-I/opt/openrov/firmware/libraries/BNO055" "-I/opt/openrov/firmware/libraries/I2C/src" "/opt/openrov/firmware/build/sketch/CMPU9150.cpp" -o "/opt/openrov/firmware/build/sketch/CMPU9150.cpp.o"
Nov 22 07:20:44 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: /opt/openrov/firmware/build/sketch/CMPU9150.cpp: In member function 'virtual void CMPU9150::Update(CCommand&)':
Nov 22 07:20:44 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: /opt/openrov/firmware/build/sketch/CMPU9150.cpp:127:29: error: 'class mpu9150::MPU9150' has no member named 'm_data'
Nov 22 07:20:44 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: /opt/openrov/firmware/build/sketch/CMPU9150.cpp:128:30: error: 'class mpu9150::MPU9150' has no member named 'm_data'
Nov 22 07:20:44 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: Using library EEPROM at version 2.0 in folder: /opt/openrov/arduino/hardware/openrov/avr/libraries/EEPROM
Nov 22 07:20:44 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: exit status 1
Nov 22 07:20:45 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: Build failed
Nov 22 07:20:45 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: StateMachine error: `arduino-builder -verbose -compile -warnings all -build-path /opt/openrov/firmware/build -hardware /opt/openrov/arduino/hardware -tools /opt/openrov/arduino/hardware/tools -libraries /opt/openrov/arduino/hardware/openrov/avr/libraries -libraries /opt/openrov/firmware/libraries -fqbn openrov:avr:mega:cpu=atmega2560 /opt/openrov/firmware/staged/OpenROV2x` failed with code 1
Nov 22 07:20:45 OpenROV orov-cockpit[661]: FIRMWARE UPDATE: Firmware update failed!

BrianAdams, 2016-11-22:
It appears this checkin: https://github.com/OpenROV/openrov-software-arduino/commit/5ab58ec54acff90c19ec751ec2b5d184d326d419 broke the firmware build.

spiderkeys, 2016-11-22:
Should be fixed now. My bad. I copied the same pattern from the BNO sensor for the zeroing commands but forgot that the MPU structure was slightly different.

BrianAdams, 2016-11-22:
Confirmed fixed

closed 2016-11-22 by BrianAdams