OpenROV/openrov-software #581: npm and Mock dev install - failed npm run deploy:prod

binary42, 2016-12-05:
When running : npm run deploy:prod

Following output:

> OpenROV-Cockpit@31.0.0 deploy:prod ../openrov-cockpit
> env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install

npm WARN package.json OpenROV-Cockpit@31.0.0 No license field.
/
> serialport@5.0.0-beta1 install ../openrov-cockpit/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "../openrov-cockpit/node_modules/serialport/build/Release/serialport.node" is installed via remote

> fibers@1.0.15 install ../openrov-cockpit/node_modules/fibers
> node build.js || nodejs build.js

`linux-x64-46` exists; testing
Binary is fine; exiting

> OpenROV-Cockpit@31.0.0 install../openrov-cockpit
> node install_lib/bower_install_all.js && node install_lib/npm_install_all.js

Ingnoring ../openrov-cockpit/.git
Ingnoring ../openrov-cockpit/node_modules
Execute bower install on ../openrov-cockpit/webcomponent_docs/bower.json
../openrov-cockpit/install_lib/bower_install_all.js:13
    if (!bowersToInstall.includes(file.substring(0, file.lastIndexOf('/')))){
                         ^

TypeError: bowersToInstall.includes is not a function
    at EventEmitter.<anonymous> (../openrov-cockpit/install_lib/bower_install_all.js:13:26)
    at emitTwo (events.js:87:13)
    at EventEmitter.emit (events.js:172:7)
    at onstat (../openrov-cockpit/node_modules/findit/index.js:182:21)
    at ../openrov-cockpit/node_modules/findit/index.js:133:22
    at FSReqWrap.oncomplete (fs.js:82:15)

npm ERR! Linux 3.16.0-38-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v4.6.0
npm ERR! npm  v2.15.9
npm ERR! code ELIFECYCLE
npm ERR! OpenROV-Cockpit@31.0.0 install: `node install_lib/bower_install_all.js && node install_lib/npm_install_all.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the OpenROV-Cockpit@31.0.0 install script 'node install_lib/bower_install_all.js && node install_lib/npm_install_all.js'.
npm ERR! This is most likely a problem with the OpenROV-Cockpit package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install_lib/bower_install_all.js && node install_lib/npm_install_all.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs OpenROV-Cockpit
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls OpenROV-Cockpit
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     ../openrov-cockpit/npm-debug.log

npm ERR! Linux 3.16.0-38-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "run" "deploy:prod"
npm ERR! node v4.6.0
npm ERR! npm  v2.15.9
npm ERR! code ELIFECYCLE
npm ERR! OpenROV-Cockpit@31.0.0 deploy:prod: `env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the OpenROV-Cockpit@31.0.0 deploy:prod script 'env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install'.
npm ERR! This is most likely a problem with the OpenROV-Cockpit package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs OpenROV-Cockpit
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls OpenROV-Cockpit
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     ../openrov-cockpit/npm-debug.log

binary42, 2016-12-05:
Looks like bowersToInstall.includes -> includes is not recognized as a function.

simonech, 2016-12-05:
Which version of node/npm do you have?
I had the same error, but then I upgraded to latest version of Node LTS (6.9.1) and now I don't get the error any more

binary42, 2016-12-05:
npm version 4.0.3 node version 7.2.0:

However, once I tried the updates for both npm and node again, I now get a problem with git connections:

> OpenROV-Cockpit@31.0.0 deploy:prod /home/borotech/Dev/repos/openrov-cockpit
> env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install

npm ERR! git clone --template=/home/borotech/.npm/_git-remotes/_templates --mirror git://github.com/plietar/node-systemd.git /home/borotech/.npm/_git-remotes/git-github-com-plietar-node-systemd-git-eaa4f9c51f56bdcaaebffb865d1d53fb7585651e-f34e9855: Cloning into bare repository '/home/borotech/.npm/_git-remotes/git-github-com-plietar-node-systemd-git-eaa4f9c51f56bdcaaebffb865d1d53fb7585651e-f34e9855'...

etc.  

something is a bit awry with my system I suspect. I will keep this ticket up and open until I get it, though. 

Thanks!

Jim

BrianAdams, 2016-12-08:
There is also this report from the forums.  

This is possible a Node7 in combination with the frozen version of the dependencies we are using.  Couple options:

https://forum.openrov.com/t/mock-dev-enviroment-for-desktops/5043/5

binary42, 2016-12-09:
I managed to get a solution running: node v6.9.2 with npm v3.10.9 Linux kernel 3.16.0-38

My first few attempts had node v4.... and subsequent node v7... node 6.9.2 did the trick... ugh.

Thanks

Jim

closed 2016-12-09 by binary42