OpenROV/openrov-software #583: Installing via NPM fails

simonech, 2016-12-06:
I've been trying to install following the [Dev How to doc](https://github.com/OpenROV/openrov-cockpit/blob/master/docs/DEV-HOWTO.md).

It mentions that the `goemux` dependency should fail on Intel machines (running on Mac), but that install process should go on.

In my case it does not... and the npm process ends with an error.

simonech, 2016-12-06:
I tried removing the goemux plugin and npm reaches the end of the without errors, so I guess the problem is that errors in NPM while installing plugins should be ignored.

BrianAdams, 2016-12-08:
@simonech can you post the npm error report?  The geomux-video-server is marked as an optional dependency and thus should not under a normal install cause the install to fail.  That said, npm-shrinkwrap has shown an inclination to break those rules.  

Will work on recreating

simonech, 2016-12-10:
Tried again. Here what I get

In the console:

```
======== cleaning =======
/Users/simone/Projects/openrov-cockpit/src/plugins/geomuxp/node_modules
======== installing =======
/Users/simone/Projects/openrov-cockpit/src/plugins/geomuxp

child_process.js:526
    throw err;
    ^

Error: Command failed: npm install

    at checkExecSyncError (child_process.js:483:13)
    at execSync (child_process.js:523:13)
    at /Users/simone/Projects/openrov-cockpit/install_lib/npm_install_all.js:56:17
    at next (/Users/simone/Projects/openrov-cockpit/node_modules/rimraf/rimraf.js:74:7)
    at CB (/Users/simone/Projects/openrov-cockpit/node_modules/rimraf/rimraf.js:110:9)
    at /Users/simone/Projects/openrov-cockpit/node_modules/rimraf/rimraf.js:136:14
    at FSReqWrap.oncomplete (fs.js:123:15)

npm WARN OpenROV-Cockpit@31.0.0 No license field.
npm ERR! Darwin 15.6.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v6.7.0
npm ERR! npm  v4.0.5
npm ERR! code ELIFECYCLE
npm ERR! OpenROV-Cockpit@31.0.0 install: `node install_lib/bower_install_all.js && node install_lib/npm_install_all.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the OpenROV-Cockpit@31.0.0 install script 'node install_lib/bower_install_all.js && node install_lib/npm_install_all.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the OpenROV-Cockpit package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install_lib/bower_install_all.js && node install_lib/npm_install_all.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs OpenROV-Cockpit
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls OpenROV-Cockpit
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/simone/Projects/openrov-cockpit/npm-debug.log

npm ERR! Darwin 15.6.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "run" "deploy:prod"
npm ERR! node v6.7.0
npm ERR! npm  v4.0.5
npm ERR! code ELIFECYCLE
npm ERR! OpenROV-Cockpit@31.0.0 deploy:prod: `env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the OpenROV-Cockpit@31.0.0 deploy:prod script 'env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the OpenROV-Cockpit package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     env 'npm_config_unsafe_perm=true' env 'npm_config_shrinkwrap=true' NODE_ENV=production npm install
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs OpenROV-Cockpit
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls OpenROV-Cockpit
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/simone/Projects/openrov-cockpit/npm-debug.log
```

then I have 2 npm-debug.log files, one in the root (which doesn't say much, basically same as the console 
[root_npm-debug_log.txt](https://github.com/OpenROV/openrov-software/files/643786/root_npm-debug_log.txt)
and one in the `openrov-cockpit/src/plugins/geomuxp` folder 
[geomux_npm-debug_log.txt](https://github.com/OpenROV/openrov-software/files/643789/geomux_npm-debug_log.txt) which seems to fail when installing zmq libe and rebuilding `zmq@2.15.3~install: Args: [ '-c', 'node-gyp rebuild' ]`

BrianAdams, 2016-12-11:
We made some changes last night. node-gyp was having a fit with linux with one of the environmental variables.  May fix it for you as well, though on my Mac I am having a heck of a time reproducing the issue.