OpenROV/openrov-software #584: No video running on Mock

simonech, 2016-12-06:
I followed the instructions on the [dev How to](https://github.com/OpenROV/openrov-cockpit/blob/master/docs/DEV-HOWTO.md). I get an error when installing via npm #583 

After this, I can run the cockpit (via VSCode), but I get a shitload of exception in the debug console.
One of them is about the `mjpeg-video-server` not being found. It might be because the previous error prevented the all remaining plugins to be downloaded?

I downloaded the ffmpeg lib.. it should have been enough, right?

simonech, 2016-12-06:
Depends of #583 (mock video works if I disable the plugin that fails

BrianAdams, 2016-12-08:
Closing given this is a side effect of an incomplete install due to #583

closed 2016-12-08 by BrianAdams