OpenROV/openrov-software #585: NPM `deploy:dev` fails

simonech, 2016-12-06:
I've been trying the various install procedures explained in [DEV-HOWTO](https://github.com/OpenROV/openrov-cockpit/blob/master/docs/DEV-HOWTO.md) and while launching `npm run deploy:prod` works fine, the `deploy:dev` fails.

It looks like it fails while installing the `phantom` packages inside the `system` folder. The error says `try running the command as root/administrator`.

But if I launch it with `sudo` it doesn't go at all.

BrianAdams, 2016-12-08:
Thanks, the npm packages have been a bit flaky on the install requirements per platform.  I am going to simply remove phantom as per the notice in a prior issue to see if that clears things up.

BrianAdams, 2016-12-08:
https://github.com/OpenROV/openrov-cockpit/commit/62a0a5823b004a5e8d8e168fb6c6eda07a911da6 should have fixed the reported issue.

simonech, 2016-12-08:
Will test again and let you know

simonech, 2016-12-10:
Tried again, but still doesn't install.
I get this 
[npm-debug.log](https://github.com/OpenROV/openrov-software/files/643792/npm-debug_log.txt) but doesn't say much

jackyea, 2017-04-21:
While I typed ''npm run deploy:prod,I got this:
 
[image]

Anybody knows why?

aalmiramolla, 2017-05-28:
I probe on master, and error too, like @jackyea 

Anyone knows how to fix this? It's my first time using node and I don't know how to solve this. I'll try changing node/npm versions and hope that that solve the problem.
Also, when I launch cockpit.js, a core dumped occurs
```
$> USE_MOCK=true HARDWARE_MOCK=true configfile='/tmp/rovconfig.json' pluginsDownloadDirectory='/tmp/plugins' node src/cockpit.js
{"pid":15086,"hostname":"Thinkpad-T440s","name":"main","level":50,"time":1496002116603,"msg":"PLATFORM: Failed to load platform details for this system: Failed to load platform name","v":1}
{"pid":15086,"hostname":"Thinkpad-T440s","name":"main","level":50,"time":1496002116604,"msg":"Error starting plugins: ","type":"Error","stack":"TypeError: Cannot set property 'name' of null\n    at Promise.try.then (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/src/lib/PluginLoader.js:62:37)\n    at tryCatcher (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:512:31)\n    at Promise._settlePromise (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:569:18)\n    at Promise._settlePromise0 (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:614:10)\n    at Promise._settlePromises (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:693:18)\n    at Promise._fulfill (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:638:18)\n    at Promise._resolveCallback (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:432:57)\n    at Promise._settlePromiseFromHandler (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:524:17)\n    at Promise._settlePromise (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:569:18)\n    at Promise._settlePromise0 (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:614:10)\n    at Promise._settlePromises (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/promise.js:689:18)\n    at Async._drainQueue (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/async.js:133:16)\n    at Async._drainQueues (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/async.js:143:10)\n    at Immediate.Async.drainQueues (/home/aalmiramolla/dev/git/TFM/rov-software/openrov-cockpit/node_modules/bluebird/js/release/async.js:17:14)\n    at runCallback (timers.js:672:20)","v":1}
 1: node::Abort() [node]
 2: 0xfafda9 [node]
 3: v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&)) [node]
 4: 0x80df1a [node]
 5: 0x80e3e8 [node]
 6: 0x2dc9f44843a7
Aborted (core dumped)
```

BrianAdams, 2017-05-29:
@jackyea and @aalmiramolla , what version of node is installed?

aalmiramolla, 2017-05-29:
I tried with various node versions,
- v6.9.2
- v6.10.3
- v7.10.0