OpenROV/openrov-software #587: NPM install fails on Windows with production settings

simonech, 2016-12-14:
Installing on Windows fails with the production settings:

On the console I have
```
======== installing =======
C:\Users\user\Documents\GitHub\openrov-cockpit\src\static
undefined
done processing plugin install
======== cleaning =======
/bower_components
======== installing =======

undefined
No bower.json present

npm WARN optional Skipping failed optional dependency //fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.0.15
npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.0.15
npm WARN OpenROV-Cockpit@31.0.0 No license field.
npm ERR! Windows_NT 10.0.10586
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\user\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.9.1
npm ERR! npm  v3.8.8
npm ERR! code ELIFECYCLE
npm ERR! OpenROV-Cockpit@31.0.0 install: `node install_lib/bower_install_all.js && node install_lib/npm_install_all.js`
npm ERR! Exit status 1
```
Here is the [npm-debug.log](https://github.com/OpenROV/openrov-software/files/652004/npm-debug.log.txt)

mmehr2, 2017-10-12:
Same thing here. I didn't do verbose logging, but tried both settings and neither would work.

At first, it seemed that the problem was with Python (I had Python 3.6 installed and it was in the path).
The error messages indicated that node-gyp required Python < 3.0 and that I could set the PYTHON env variable to point to it. So I also installed Python 2.7, removed 3.6 from the path, and added 2.7 using that variable.

Then it complained that gyp was requiring .NET framework 2.0 (Visual C++ 2005) to rebuild itself. That is so obsolete it's not even funny. 

So, does this mean I am stuck and cannot install and run the MOCK versions of the code with the latest tools? Any suggestions would be greatly appreciated!

Regards,
Mike
[npm-debug.log](https://github.com/OpenROV/openrov-software/files/1380955/npm-debug.log)

mmehr2, 2017-10-13:
UPDATE: I found out that there is now a way to add the Microsoft windows-build-tools using npm here:
https://www.npmjs.com/package/windows-build-tools
This will automatically perform another install of Python 2.7 as well as Visual C++ 2015 and SDKs and configure the npm system to use it.

So now the problem is that it can't find an include file when compiling:
C:\Users\Mike\Documents\Projects\openrov\openrov-cockpit\node_modules\i2c>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  i2c.cc
  win_delay_load_hook.cc
..\src\i2c.cc(7): fatal error C1083: Cannot open include file: 'sys/ioctl.h': No such file or directory [C:\Users\Mike\Documents\Projects\openrov\openrov-cockpit\node_modules\i2c\build\i2c.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Mike\Documents\Projects\openrov\openrov-cockpit\node_modules\i2c
gyp ERR! node -v v6.11.4
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

So it can't find sys/ioctl.h when compiling ../src/i2c.cc which I can't find in the Cockpit source files either.

I did notice that the windows-build-tools install won't install the .NET Framework 4.5.1, but when I try to install it, it says it is already installed on my system. (Where?)

If I learn anything more on getting node-gyp to compile on my system (and hopefully then the rest of the system builds, but I'm not too optimistic!), I'll post again.

Regards,
Mike
[npm-debug.log](https://github.com/OpenROV/openrov-software/files/1381112/npm-debug.log)