OpenROV/openrov-software #600: Need to add a battery preset for the NMC batteries

BrianAdams, 2016-12-28:


BrianAdams, 2016-12-28:
@briangrau , these are just for Trident, correct?

spiderkeys, 2016-12-28:
If this is for Trident, then this plugin won't be employed. The battery management IC provides all of the processed information and telemetry relating to the batteries.

If this is for the 2X, let me know and I can add a new battery class.

briangrau, 2016-12-31:
This is for the 2-series. These are the blue batteries (the ones we used in the Deep Units). These are in the store and so the battery configuration should be in the software as well. You just need the upper and lower voltage limits?

spiderkeys, 2017-01-02:
Please advise the voltages and I can wrap this one up :)

spiderkeys, 2017-01-02:
Fixed in pending PR: https://github.com/OpenROV/openrov-cockpit/pull/183