OpenROV/openrov-software #607: Pattern: Enable/Disable of plugins

BrianAdams, 2017-01-05:
In progress...

We have a single logical application that we are deploying across multiple ROVs products.  We want the system to gracefully degrade functionality if the platform that it is deployed on is unable to service features available in the application.  Ex, if the system does not have lights, we should not show buttons and options for enabling/disabling lights.

There are a couple of use cases to use to rationalize the design:
1. Making available the plugins based on the application, cockpit or dropcam
2. The ROV has or does not have lights
3. The IMU which was sending accelerometer data switches modes and no longer sends that same data.
4. Given that a plugin can be spread over multiple layers of the application, how does the entire system enable/disable together

Assumptions:
* Our entire system is built on top of decoupled messaging.  We should continue to leverage that pattern as much as possible.
* If we don't have to enable a plugin because it is not needed on a platform, it would be good not to load it.

Prior Art:
* https://github.com/OpenROV/openrov-software/issues/555

[image]
[Edit](https://www.websequencediagrams.com/#open=199712)

spiderkeys, 2017-01-05:
Some observations on approaches to implementing this, assuming the pub/sub routing table is in place:

- Choose a message definition format to use system wide (.idl, .proto, etc)
  - Protobufs is a good candidate to start with as we have support for it in all of our employed languages
- Create an MCU interface service (can be written in node, c++, whatever)
  - Should be started by Cockpit as a service
  - Can be networked to Cockpit with websockets, zeromq, etc
- Leave the comm interface to the FW the same for right now. I.e., leave it as "field:value;" messages
- Have the MCU interface map the FW API to our agreed upon message protocol
- After this is working, we can go back whenever we like and rev the FW<->MCU comm protocols since they are private and external to the pub/sub system in cockpit.

spiderkeys, 2017-01-18:
Proposed message format:

```ts
class ROVHeader
{
   status: number; // 0 - Disposed, 1 - Active, 2 - Inactive
}

class ROVMessage
{
   topic: string;
   header: ROVHeader;
   payload: any;
}
```

From the firmware's point of view, it will treat individual sensors as a group of topics. To use the IMU as an example with the following state messages:

IMU.State = { id: number, statusCode: number }
IMU.Fused = { id: number, roll: number, pitch: number, yaw: number }
IMU.Raw = { id: number, mag[ 3 ]: number, acc[ 3 ] : number, gyro[ 3 ] : number }
IMU.Mode = { id: number, mode: number }
IMU.Cal = { id: number, system: number, acc: number, gyro: number, mag: number }

```c++
// Choose an ID
m_id = 3;

// Register the IMU by sending a state message to say we're alive
SERIAL_PRINTF( "is:%d|%d", m_id, EStatus::ALIVE );

// Check for new message
auto msg = NComms::GetMessage();

// Listen for mode switch command (  "imr": "IMU Mode Request" )
if( msg->IsType( "imr" ) && msg->arguments[ 0 ] == m_id  )
{
   // Get requested mode
   int requestedMode = msg->arguments[ 1 ];

   // Send target mode as an acknowledgement ( "imt": "IMU Mode Target" )
   // Sending a negative value could represent an error code. Contextual decision to make
   SERIAL_PRINTF( "imt:%d|%d", m_id, requestedMode );

   // Do a bunch of things that eventually gets us to that mode

   // Whenever a mode change occurs, print the current mode ( "im": "IMU Current Mode" )
   SERIAL_PRINTF( "im:%d|%d", m_id, m_mode );
}
```

On the MCU manager side, classes can be used to model the devices as implemented on the firmware side. Serialization between the MCU and MCU Interface is a separate concern from the serialization (if any) used on the cockpit bus, but the message information should be translatable between the two.

The MCU interface will translate firmware information into "ROVMessage" instances as described by the Cockpit system bus and handle publishing/subscribing to the available API that the firmware implements. Eventually the MCU interface should also be responsible for best effort reliability in terms of transport over the wire to the MCU. This could be a separate design discussion.