OpenROV/openrov-software #615: input-controller prefix code causing crash on image

BrianAdams, 2017-01-12:
```
Jan 11 21:15:50 openrov orov-cockpit[698]: /opt/openrov/cockpit/src/system-plugins/input-controller/index.js:26
Jan 11 21:15:50 openrov orov-cockpit[698]:         files.forEach(file => {
Jan 11 21:15:50 openrov orov-cockpit[698]:              ^
Jan 11 21:15:50 openrov orov-cockpit[698]: TypeError: Cannot read property 'forEach' of undefined
Jan 11 21:15:50 openrov orov-cockpit[698]:     at fs.readdir (/opt/openrov/cockpit/src/system-plugins/input-controller/index.js:26:14)
Jan 11 21:15:50 openrov orov-cockpit[698]:     at FSReqWrap.oncomplete (fs.js:123:15)
```

nerdgilbert, 2017-01-12:
Looks like the code to load custom presets. Did I put the file path for the directory wrong (`./src/static/presets/`)? Does it change on the image?

BrianAdams, 2017-01-12:
possibly. I'm debugging right now to see what it is using as the base for the relative path.

closed 2017-01-12 by BrianAdams

nerdgilbert, 2017-01-12:
Good catch @BrianAdams :+1: