OpenROV/openrov-software #623: The system appears to be sending multiple commands on keyboard press

BrianAdams, 2017-01-19:
For instance, the hitting light will toggle it off-on-off-on.

BrianAdams, 2017-01-19:
Checked the input manager to ensure it was not somehow holding on to listeners. Registered several different presets, including the same one a couple time. In mock mode, sending a light command only happened once.

nerdgilbert, 2017-01-19:
Was this reported on a vehicle?

BrianAdams, 2017-01-19:
@nerdgilbert , yes. The test vehicle Grau was using.  If you can get a chance to take a look on vehicle, I think that is the next step.  Might be worth capturing the serial traffic while recreating the issue.

BrianAdams, 2017-01-19:
@spiderkeys had a sense it might have to do with the syncTargetPower function that keeps retrying every 33ms until it gets a lights_tpow response from the MCU.  That it may be the round trip is taking too long and the message gets sent again. Now how a power off message would trigger a power on again is still a mystery.

spiderkeys, 2017-01-20:
The command going to the firmware is a specific value. The only place the concept of toggling exists is the browser client, so delayed/repeated toggles are probably building up between the client and server. This behaviour has exhibited itself in many other commands as well, happening especially often on the BBB.

BrianAdams, 2017-01-20:
This was tied to the gamepad press generating multiple gamepad click events. Opened other issues. Closing this one.

closed 2017-01-20 by BrianAdams