OpenROV/openrov-software #626: System no longer wires up a gamepad that is connected after cockpit has started

BrianAdams, 2017-01-20:
Start cockpit
Plugin gamepad

Expected: full control
Actually: no control from gamepad

nerdgilbert, 2017-01-21:
The Gamepad API we have been using has been deprecated. The maintainer suggests just using the provided Web APIs

nerdgilbert, 2017-01-21:
As per your steps for recreating the bug, I am having difficulty recreating it in MOCK mode on my machine. The PS4 controller always comes up when I start cockpit and plug in the gamepad. Am I missing something?

I do realize that these steps produce a bug:
 * Start cockpit
 * **Unplug** gamepad
 * Plug in gamepad

Expected: Full control
Actually: No control

BrianAdams, 2017-01-21:
I suspect this might fix it:
```diff
diff --git a/src/static/js/libs/gamepad.js b/src/static/js/libs/gamepad.js
index 5dc89de..9225f0a 100644
--- a/src/static/js/libs/gamepad.js
+++ b/src/static/js/libs/gamepad.js
@@ -582,7 +582,13 @@
       console.log('Not supposed to go here!');
       return;  // should not happen
     }
-    if (gamepads.length !== this.gamepads.length) {
+    gamepads.nonNulllength = 0;
+    for (i = 0; i < gamepads.length; i++) {
+      if (gamepads[i]!==null){
+        gamepads.nonNulllength++;
+      }
+    }
+    if (gamepads.nonNulllength !== this.gamepads.length) {
       var gamepad, i;
       for (i = 0; i < gamepads.length; i++) {
         gamepad = gamepads[i];
```

Trying to find a gamepad to test with

closed 2017-01-23 by nerdgilbert