OpenROV/openrov-software #637: HD Camera feed not coming up

BrianAdams, 2017-08-11:
There was a report of the camer not coming up.  In the debug console the javascript confirmed that the browser was having trouble connecting to a url with geovideo0_1 or geovideo0_2.  

This is an indication that the firmware on the camera is not correct.  This can be confirmed with the command `sudo mxcam bootmode`

If that command returns `snor` that is a problem.  We want the command to return `usb`.

Contacting support will run a user through the commands to correct the issue in the field.

closed 2017-08-11 by BrianAdams

BulldogLovesCat, 2017-08-18:
hi Brian Adams, I have the same issue as mentioned by you. The "sudo mxcam bootmode" command returns "snor" instead of "usb". 

"Contacting support will run a user through the commands to correct the issue in the field."
Do you suggest to contact the openrov support team for correcting this issue? Is there any instruction that I can follow through myself? 

Thanks

BulldogLovesCat, 2017-08-18:
hi, happen to bring up the camera feed by chance and check the command "mxcam bootmode" again, it returns "snor". maybe this is the correct response when using the software image 31.0.0. 
but still struggling to figure out why the camera feed is intermittent?