OpenROV/openrov-software #639: How does Tether communication works?

BulldogLovesCat, 2017-08-19:
I am wondering how fundamentally the tether communication works? 
When we press the key (W) on keyboard, the port and starboard motors start to run.
what kind of control signal is generated and pass over the tether cable from the Topside adapter to the homeplug adapter on the ROV?

It is definitely not sending different DC voltages as I measure the voltage across the two tether cables, it gives me a constant voltage of 4.4~4.5V. So it could be a small AC signal? As I cannot visualize any changes in terms of AC voltage using a multi-meter. 

Is this some sort of power line communication stuff? 

Could you explain in both microscopic (such as EM waves, Current, RF and etc) and macroscopic (such as coding, architecture) point of view?

Thanks

scubasonar, 2017-08-19:
The homeplug adapters are normally used to give you ethernet over the same wires that feed 110V AC(in the US) across your house to the wall plugs. By splitting that adapter and removing the part that handles 110VAC you're left with a bit of electronics (in the topside adapter and ROV) that will gives you a network over two wires. 

With all the physical stuff wrapped up by a homeplug you're just connecting to a server like on your desk.

BulldogLovesCat, 2017-08-20:
@scubasonar  Thank you for your reply. 

" a bit of electronics (in the topside adapter and ROV) that will gives you a network over two wires." 
*************************************************************************************************************
How does it work? The cat5 cable has 4 wires (Transmit+/- and Receive+/-) and the network has only two wire (Tether A and B). what kind of signal is running through the tether cable? Is it some sort of electrical pulses? Any waveform diagrams? I really would like to imagine what do those network signals look like?

BulldogLovesCat, 2017-08-20:
Could you show me the difference between the network signals for turning the internal LED on and turning the motor on?

scubasonar, 2017-10-24:
https://en.wikipedia.org/wiki/Power-line_communication . Hook one up to an oscilloscope and see what is up?

mcdafydd, 2017-11-23:
I find the Homeplug protocols to be similar to 802.11 wireless.  They are a reliable protocol on layer 2, meaning, every packet sent via a Homeplug link must be ACK'd by the adapter at the target location.  In a tether application, let's say 400 meters as that's a practical limit for the usual -50dBm/Hz FCC limits for Homeplug AV frequencies, the protocol will be seriously impacted by excessive noise in the tether.  The cleaner you can make the environment for the Homeplug signals, the lower your latency between Homeplug adapters at either end of the tether.

Homeplug adapters act as bridge devices between Ethernet (or wi-fi) and Homeplug protocol. If you want to see "how it works", use wireshark on either end of an Ethernet connection and look at the packets.  I would recommend assuming that the reliable Homeplug protocol will deliver your data.  Instead, just worry about how long it takes to get there.  If latency is much longer than distance + serialization, you may have a noise problem.  I find a typical home Ethernet link to run around 3-5ms average, where a direct Ethernet cat5 or higher connection would sit under 1ms.