OpenROV/openrov-software #645: How to upgrade my old 2.8 Rov Openrov ?

HBlanchet300, 2024-10-25:
Fine, :smiley:

Here is my problem :smiley:
I use an old 2.8 Rov OpenRov which was working with Chrome software.

I try to use it again, it works quit well but and when I use the a or q , the video stops and nothing happends after.

Nothing happends with the Joystik.

Do I have to upgrade something ?
I think Yes.
I can purchase a new SDCard with the software still inside, because I am not a specialist looking soft and hardware ?

Is somebody can help me to do my OpenRov working again ?

Thank’s

Herve France October 25st 2024    hblanchet31@gmail.com

TCIII, 2024-11-13:
@HBlanchet300,
What version of Google Chrome are you using?
Regards,
TCIII

HBlanchet300, 2024-11-13:
Hello TCII

Thank's for your answer.

The Google Chrome version is the latest (I have download just now) Version 131.0.6778.70 (Build officiel) (64 bits)

![Version cockpit OpenRov 2 8 HBlanchet]
(https://github.com/user-attachments/assets/54fb5c75-ba48-424f-afbf-11c4f6ee867e)

Version cockpit OpenRov 2.8 HBlanchet.JPG

Thank you

Herve

TCIII, 2024-11-13:
@HBlanchet300,
It is a known fact that the OpenROV Cockpit will not work any version of Google Chrome after around 2018. I have seen recommendations to fallback to Google Chrome version 79, but even that does not work for me.
Regards,
TCIII

HBlanchet300, 2024-11-18:
I have found on SlimJet site and reloaded this version 48 (64 bits) of  Chrome : 48.0.2564.97 

The joystick works well now except the Q and Z fonctions as on the keyboard (up and down camera) : the flow of the video stops.
The camera himself is well moving but the video and image (display) are stopped.

I have stopped the upgrade of Chrome and when I use my rov, I disconnect the Wifi and the internet connection options.

Is somebody can help me again to get a solution looking the A Q and Z buttons ?

Regards
@Hblanchet300

:)

TCIII, 2024-11-18:
@HBlanchet300,
I have an OpenROV 2.8 with the high resolution pro camera running OpenROVSuite_31.0.0-rc7.

I tried using Chrome version 48.0.2564.97 and here are the results:
1) No video, but the reticle is displayed in the Field of View.
2) The top banner is displayed and the run time and battery voltage is displayed below the top banner, but there is no bottom banner.
3) The keyboard controls all of the ROV functions.

Questions:
1) What version of the OpenROVSuite software are you using?
2) Do you happen to have a copy of the OpenROV 2.8 build manual or operating manual?

Regards,
TCIII

HBlanchet300, 2024-11-19:
Fine TCIII    Tuesday, November 19 th 2024    13h58   01h58 PM

Object : Re: [OpenROV/openrov-software] How to upgrade my old 2.8 Rov Openrov ? (Issue #645) - 241119 -

Here is the Chrome version I use now :
Version 49.0.2623.75 (64-bit)

So :
- The video is OK except when I use the Q or Z button on the keyboard or the Y and A buttons on the joystick (Logitech Gamepad F310).
- On the screen display, at the bottom, all works well .
    Do yo want a screen copy ?

- The keyboard controls all of the ROV functions as you except the Q and Z buttons which stop the vidéo.

Here is the OpenRovSuite which is inside the rov :
File : Screen copy version OpenRov 2_8 RV 241119.jpg

Does it what do you need ?

If not, where I can get the information ?

Looking you request : copy of the OpenROV 2.8 build manual or operating manual
I have this file :
OpenRov 2_8 guide_80_en RV 241119.pdf.jpg

Do you need it ?
I can send you it.

I had so others files BUT, the link which sends to the web site DOZUKI is no more available 😭

Ah !
Do you know any software which permit to record the videos of the Rov ?
I would be able to send you the videos of now.

Thank you

Best regards

Herve

>
[image]
[image]